Apollo3zehn/FluentModbus #2: Add events, on data change.

Apollo3zehn, 2019-05-06:


ChangeTheCode, 2021-02-08:
is there a plan when this feature will be come? Currently we have to query the register to see if anything has changed, is this correct ?

Steve0212a, 2021-02-12:
I second the need for this.  We are looking to convert our Modbus server to .net core and your library works very well.  This is the only feature it is missing that we need.

I really like your server, but without this feature, we may have to look elsewhere as we do not want to regularly poll for data changes.

Thanks for writing this server.

Apollo3zehn, 2021-02-12:
I'll check that in monday. What is your specific need? Geht notifications for single registers or for a full range of registers?

ChangeTheCode, 2021-02-14:
For my point of view at the first time it would help if one of both would be supported. So you do not have to poll each time hand have to implement something to detect if something has changed in one register.

closed 2021-02-15 by Apollo3zehn

Apollo3zehn, 2021-02-15:
I have pushed an update (3.0.1-preview.1) to nuget with the requested feature. A sample to get notifications on data change is shown here: https://apollo3zehn.github.io/FluentModbus/index.html#events

It would be great if you could test it and tell me if it suits your needs. Thanks :)

mharrand, 2021-02-19:
Hi,  I'm working with Steve0212a.   I've implemented the new events and they seem to be working.  Question though, when operating in Asynchronous mood, should I be using the server lock around the Gets and Sets in the event handlers?

Apollo3zehn, 2021-02-19:
Yes you should use lock here, too. Otherwise it could happen that you and a request handler read/write to the same buffer location. Here is the lock statement of the request handler that you should synchronize with using your own lock: https://github.com/Apollo3zehn/FluentModbus/blob/master/src/FluentModbus/Server/ModbusRequestHandler.cs#L107

commit 1b25fcfd (2021-02-15): Add events on data change (fixes #2).