Apollo3zehn/FluentModbus #8: Client: Make timeout configurable.

Apollo3zehn, 2019-05-10:


11v1, 2019-11-13:
Hello.

I need this tweak! ;-)

Apollo3zehn, 2019-11-13:
Hi, I added two properties to allow specifying the connect timeout and the read timeout. You can use it like [this](https://github.com/Apollo3zehn/FluentModbus/blob/master/tests/FluentModbus.Tests/ClientTests.cs#L24-L27) (the snipped shows only the usage of ```ConnectTimeout``` but it works the same for ```ReadTimeout```. The time is provided in milliseconds.

I uploaded a new NuGet package (https://www.nuget.org/packages/FluentModbus). I hope this works for you, if not please file a new issue.

closed 2019-11-13 by Apollo3zehn

Apollo3zehn, 2019-11-17:
I published another release with the possiblity to specify a timeout for the server. After this timeout (default: 1 minute) client connections will be closed. I found a bug that the server's timeout timer was never reset which caused the connection to be aborted after 1 minute. Unfortunately, this bug was not covered by the unit tests. 

I released version 1.4.0 with the bug solved and a unit test added to prevent that this happens again.