Apollo3zehn/FluentModbus #18: Exception trying to read more then 123 registers

11v1, 2019-11-22:
256 bytes is not enough to read all data. MBAP + PDU maximum size is 260 bytes.
```
public ModbusTcpMessageBuffer()
{
    this.Buffer = ArrayPool<byte>.Shared.Rent(256);

    this.RequestWriter = new ExtendedBinaryWriter(new MemoryStream(this.Buffer));
    this.ResponseReader = new ExtendedBinaryReader(new MemoryStream(this.Buffer));
}
```

Apollo3zehn, 2019-11-22:
According to Wikipedia, maximum message size is 
256 Bytes and 123 registers for Modbus TCP:

> Because the number of bytes for register values is 8-bit wide and maximum modbus message size is 256 bytes, only 125 registers for Modbus RTU and 123 registers for Modbus TCP can be read at once.

I'll check it later against the specification.

Apollo3zehn, 2019-11-22:
I found this explanation for different message size limits: http://wingpath.co.uk/docs/modtest/message_limits.html 

This complies with the limits in the specification I used. So, you are right that up to 260 bytes are allowed per Modbus TCP message. I will update the code this weekend. Thanks for reporting!

Apollo3zehn, 2019-11-23:
I published version 1.4.1. Can you please test it? Hopefully this solves your issue.

11v1, 2019-11-23:
I'll test it at Monday. Thanks a lot.

11v1, 2019-11-25:
Looks like it works. Thanks again!

closed 2019-11-25 by 11v1