Apollo3zehn/FluentModbus #20: Add Modbus server app for testing (WPF or Blazor+Electron?)

Apollo3zehn, 2019-12-01:


seikosantana, 2023-11-13:
Instead of WPF, Avalonia might be a good alternative considering it is cross platform.

What kind of modbus server app do you have in mind? Anything i could contribute to for this issue?

Apollo3zehn, 2023-11-13:
A replacement for https://www.modbustools.com/download.html would be nice. A tool to quickly see the Modbus server register content and be able to edit it in decimal, binary or hex mode. This tool should also be able to show the Modbus raw traffic but that is not the first requirement.

FuentModbus should be able to provide the required base functionality. Only a nice UI is missing.

Since my last two larger projects were Blazor-based, I would choose that again. I like having C# for logic / rendering and CSS for styling. 

More specific, my preferred technology stack right now is
- Blazor Server for the backend and rendering
- MudBlazor for controls
- TailwindCSS for styling (flex boxes, padding, margin, shadows, colours, ...)
- https://pictogrammers.com/library/mdi/ for Icons

However, that is only my preference. Since I do not see when I would actually be able to work on this I am happy about any type of contribution (Avalonia, ...).

So if you would like to contribute a prototype or more, please go ahead :-)

Apollo3zehn, 2023-11-14:
No, not yet unfortunately. But feel free to add such an API. I will do it eventually but that may take long until I will find the time to do so. So if you have an idea, please go ahead :-)