Apollo3zehn/FluentModbus #32: Additional features for ModbusTcpServer

mkroesen, 2020-08-25:
I have some questions about new features for the server.

Is it possible to limit the number of clients? Something like:
```csharp
server = new ModbusTcpServer();
server.MaxConnections = 5;
server.Start();
```

Is it possible to disable or to set the max addresses `server.MaxInputRegisterAddress = 10 (not only get)` for some registers?
For example:
```
- Input Register
- Holding Register
- Read Coils
...
```
So that you get an "**Illegal function**"-Error in ModbusPoll or something.

Would be great if there was a solution for this.

Apollo3zehn, 2020-08-26:
Yes that sounds like a good extension (both suggestions). I will prepare a draft one the other issue with the endiannes is clarified.

Apollo3zehn, 2020-08-31:
I have prepared the draft pull request (#33) but I am not yet done with the documentation updates.
I had to implement everything a bit different than suggested:

- `server.MaxConnections`
The property has been added like you suggested. But unfortunately the internal `TcpListener` accepts all incoming connections automatically. When the limit (MaxConnections) is exceeded, the connection is disconnected immediately. In practice this means for the client that it can connect successfully but will fail trying to execute any Modbus functions with an `IOException`. See test here (https://github.com/Apollo3zehn/FluentModbus/blob/feature/server-extensions/tests/FluentModbus.Tests/ModbusTcpServerTests.cs#L104-L142).

- `server.MaxInputRegisterAddress`
This has been implemented differently to allow much more powerful checks of incoming client requests. For this I added an `RequestValidator` property which takes an action that is executed whenever a client requests something. With this you are able to fine tune your register min and max limits for each register type and can also return the Modbus exception code you want. See this test for an example: https://github.com/Apollo3zehn/FluentModbus/blob/feature/server-extensions/tests/FluentModbus.Tests/ModbusServerTests.cs#L107-L186

- Big Endian issues:
Unfortuantely I have overseen that in the extension method I planned to implement there is no way to access the server`s constructor arguments.

Thus I added two new methods: `Span<short> GetInputRegisters()` and `Span<short> GetHoldingRegisters()`. The return value is a `Span<short>` and there are several extension methods for this type. These are:

- `void SetValueLittleEndian<T>(this Span<short> buffer, ushort startingAddress, T value)`
- `void SetValueBigEndian<T>(this Span<short> buffer, ushort startingAddress, T value)`
- `T GetValueLittleEndian<T>(this Span<short> buffer, ushort startingAddress)`
- `T GetValueBigEndian<T>(this Span<short> buffer, ushort startingAddress)`

An example usage can be found here: https://github.com/Apollo3zehn/FluentModbus/blob/feature/server-extensions/tests/FluentModbus.Tests/ModbusServerTests.cs#L47-L71

The advantage with `Span<short>` is that I don't need any `startOffset` property anymore.

The performance is quite good for my understanding:

[image]

[image]

I´ll finish this tomorrow.

mkroesen, 2020-09-01:
That sounds great.
I appreciate your changes then.
Will you also update the nuget package?
Thanks.

Apollo3zehn, 2020-09-01:
The update is finished now (there have been minor naming changes):

https://github.com/Apollo3zehn/FluentModbus/releases/tag/v2.2.0
The nuget package is also updated: https://www.nuget.org/packages/FluentModbus/

I hope this solves your issues and fits to your needs. If not, please come back to discuss further modifications.

closed 2020-09-01 by Apollo3zehn