Apollo3zehn/FluentModbus #35: ModbusTcpServer.MaxConnections not working

mkroesen, 2020-09-18:
I have set `MaxConnections = 2` but more connections can be established without problems and I also receive data.

[image]

Have I configured something wrong?
I would be happy about a solution.

Apollo3zehn, 2020-09-18:
I will also check that in Monday :-) in my tests it is working but I test again with your setup/client.

Apollo3zehn, 2020-09-21:
There was a wrong *greater than* / *greater equal* check and I used the `lock` keyword incorrectly which allowed the connection of more than the limited number of clients when they connect at the same time. I am not sure if that solves your issue but in my tests it worked (tested with Modbus Poll as you did). The found issues have been released in version 2.2.1 (https://www.nuget.org/packages/FluentModbus/2.2.1).

Apollo3zehn, 2020-09-21:
It looks like that you are creating multiple connection in the same Modbus Poll application (4 child windows in one main window). Maybe Modbus Poll is reusing the connection. I have tested it with four independent Modbus Poll windows.

mkroesen, 2020-09-22:
It works now but I can always use one more connection than indicated. If `MaxConnections = 2`, I can use 3 connections, `MaxConnections = 2` I am able to use 4 etc.

Apollo3zehn, 2020-11-06:
I have found the reason why one connection more than expected is accepted and will release a fixed version soon.

commit beedb885 (2020-11-06): Fix #35.
closed 2020-11-06 by Apollo3zehn