Apollo3zehn/FluentModbus #45: Mid Little endian

Adhara3, 2020-12-05:
Hi,

I just wanted to let you know that the server we are working on in MidLittleEndian. 
No problem, I just read bytes and then interpret byte order myself, I just wanted to let you know that there are these scenarios that are currently uncovered by the lib.

Cheers
A

Apollo3zehn, 2020-12-07:
Thanks for that hint. I have no experience with any non-Little endian systems. Only big-endian is sometimes required because network procotols often demand it.

Could you please give a short example of those numbers would look like? With the new intrinsics support of .NET Core for Intel CPUs, I can now hardware accelerate the byte shuffling. I have already done it in another project (e.g. here for AVX2: https://github.com/Apollo3zehn/HDF5.NET/blob/master/src/HDF5.NET/Filters/EndiannessConverterAvx2.cs), so it won't take too much time to implement it here.

Adhara3, 2020-12-07:
Hi,

I had myself to look at the raw bytes I was receiving from the server, then inserted them in here:
https://www.scadacore.com/tools/programming-calculators/online-hex-converter/

I knew I was expecting a particular `uint` value of 141... so I just found out which was the endianness that was giving the right number. That's how I discovered the existence of mid-little endian.
The byte layout is indicated above the boxes.

A

Apollo3zehn, 2020-12-07:
Thanks for your input, I will go through to it soon and see if I can improvide the code accordingly!

PR #53 (2021-09-21): Multi Unit RTU implementation
Hello @Apollo3zehn! 

The company I work for (Ondo Solutions) found your library quite useful and indispensable to our use cases.
However, we found out few missing  things that are critical for us, namely multiple unit support for RTU Modbus.
So, we went ahead and added support for it in a separate class called `MultiUnitRtuServer`  without moddifying the original RTU server. 
In the process we also found missing support for` MidLittleEndian` encoding which we also added just today so it resolves #45 

There are few things to polish out, but i think the current st [... cut]