Apollo3zehn/FluentModbus #51: feat: support start Modbus tcp server with an specific port

ChangeTheCode, 2021-08-22:
It would be great if the `ModbusTcpServer.Start()` also supports an function with a port configuration. I already found the `Start(IPEndPoint)` where it would be possible to set the port. 

For my point of view would be nice to have an method with for example an nullable string. If the string is null the default port 502 will be used.

Apollo3zehn, 2021-08-24:
Do you mean something like `server.Start(string? localEndpoint)`?
I am not happy with a nullable string, I would not consider this good API design. I would prefer the following:

```cs
server.Start("127.0.0.1:1024") // listens on 127.0.0.1:1024
server.Start("192.168.0.2") // listens on 192.168.0.2:502
server.Start("localhost") // listens on 127.0.0.1:502
server.Start() // listens on 127.0.0.1:502
```

Would this fit to your usecase?

ChangeTheCode, 2021-08-24:
@Apollo3zehn  I would also be happy with for e.g. `server.Start("127.0.0.1:1024") // listens on 127.0.0.1:1024` 

I understand your point about API design. So would be better to go with the approach you mentioned 👍

closed 2021-09-14 by Apollo3zehn

rligocki, 2024-04-29:
Is this feature implemented? I searched source code, but did not found any way how to specify port on which modbus TCP server should start. Commit that closed this issue changes only modbus client connect method. Thank you.

Apollo3zehn, 2024-04-29:
You can use the `void Start(IPEndPoint localEndpoint)` overload and specify the port like this: `new IPEndPoint(ipAddress, 502)`. Or do you need the exact feature of being able to pass a string to the `Start()` method like above? Something like `127.0.0.1:1024` or `0.0.0.0:1024`? I am not sure why I did not implement it for the server as well at that time but right know I would like to figure out if you need any way to specify the port or the specific way of this issue by using a string.

rligocki, 2024-04-29:
No, there is no need to pass string like this `127.0.0.1:1024`. I need just some way to specify port when starting modbus TCP server. Will try method overload you mentioned. ;) Thank you. PS: IP address should be 0.0.0.0 in case I need modbus server to listen to any IP?

Apollo3zehn, 2024-04-29:
Yes, in that case it is `0.0.0.0`.

commit cf9bd65d (2021-09-14): Add new Modbus TCP client constructors. Fixes #51.