Apollo3zehn/FluentModbus #55: Timeout Exception on Linux in 4.0.0-preview1

aviatrix, 2021-10-01:
When testing the preview1 version of the package, we encountered a bug where the client times out after first read from RTU server, which was no the behavior we had on the code of my version of the PR, so this makes me thing something broke in the new version, but i don't see anything that could break it and just on Linux for that matter. 

Any help or hints would be appreciated

Apollo3zehn, 2021-10-01:
Thanks for your feedback! I'll test it on Monday when I have access to a Linux machine.

maikebing, 2021-10-01:
The Parity  Default is Parity.Even , some device don't support it , so we get a  Timeout  Exception !
if modify Parity.Even    to  Parity.None ,  we get Modbus_NoParityRequiresTwoStopBits 
so  remove  
  if (this.Parity == Parity.None && this.StopBits != StopBits.Two)
                throw new InvalidOperationException(ErrorMessage.Modbus_NoParityRequiresTwoStopBits);

[](https://github.com/Apollo3zehn/FluentModbus/pull/56)

Apollo3zehn, 2021-10-19:
According to the spec (https://www.modbus.org/docs/Modbus_over_serial_line_V1_02.pdf), section 2.5.1 RTU Transmission Mode:

- "The default parity mode **must** be even parity." 
- "... the use of no parity requires 2 stop bits."

So I think the devices are not working according to the Modbus specification. I will remove the error message to allow compatibility, but the default of the parity mode should be kept `even` to stay in line with the spec.