Apollo3zehn/FluentModbus #62: Error running .NET Core on Raspberry Pi

ciramconsulting, 2022-02-11:
Hi, we created a sample app for reading modbus using TCP for Solar Inverter. On Windows it works 100% not on Raspberry Pi .
Generated error: 
"A fatal error was encountered. The library 'libhostpolicy.so' required to execute the application was not found in '/home/pi/Ciram/'.
Failed to run as a self-contained app. If this should be a framework-dependent app, add the /home/pi/Ciram/ModbusApp.runtimeconfig.json file specifying the appropriate framework."

ModbusApp.runtimeconfig.json contains: 

{
  "runtimeOptions": {
    "tfm": "netcoreapp3.1",
    "framework": {
      "name": "Microsoft.NETCore.App",
      "version": "3.1.0"
    }
  }
}

 Other App's that we create with the same code, but without FluentModbus works on the Raspberry Pi. 

This is what we used for all our products for windows and Raspberry Pi
<Project Sdk="Microsoft.NET.Sdk">
  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp3.1</TargetFramework>
    <StartupObject>ModbusApp.Program</StartupObject>
  </PropertyGroup>
  <ItemGroup>
    <PackageReference Include="ConsoleTables" Version="2.4.2" />
    <PackageReference Include="FluentModbus" Version="3.2.1" /> -> this is new now.
    <PackageReference Include="Newtonsoft.Json" Version="13.0.1" />
  </ItemGroup>

</Project>

[image]

ciramconsulting, 2022-02-11:
If we add a full depency file (same issue)

{
  "runtimeTarget": {
    "name": ".NETCoreApp,Version=v3.1",
    "signature": ""
  },
  "compilationOptions": {},
  "targets": {
    ".NETCoreApp,Version=v3.1": {
      "ModbusApp/1.0.0": {
        "dependencies": {
          "ConsoleTables": "2.4.2",
          "FluentModbus": "3.2.1",
          "Newtonsoft.Json": "13.0.1"
        },
        "runtime": {
          "ModbusApp.dll": {}
        }
      },
      "ConsoleTables/2.4.2": {
        "dependencies": {
          "System.Reflection.TypeExtensions": "4.3.0"
        },
        "runtime": {
          "lib/netcoreapp3.1/ConsoleTables.dll": {
            "assemblyVersion": "2.0.0.0",
            "fileVersion": "2.4.2.0"
          }
        }
      },
      "FluentModbus/3.2.1": {
        "dependencies": {
          "Microsoft.Extensions.Logging.Abstractions": "3.0.0",
          "System.IO.Ports": "4.6.0",
          "System.Runtime.CompilerServices.Unsafe": "4.7.1"
        },
        "runtime": {
          "lib/netstandard2.1/FluentModbus.dll": {
            "assemblyVersion": "3.0.0.0",
            "fileVersion": "3.2.1.0"
          }
        }
      },
      "Microsoft.Extensions.Logging.Abstractions/3.0.0": {
        "runtime": {
          "lib/netstandard2.0/Microsoft.Extensions.Logging.Abstractions.dll": {
            "assemblyVersion": "3.0.0.0",
            "fileVersion": "3.0.19.46305"
          }
        }
      },
      "Microsoft.NETCore.Platforms/3.0.0": {},
      "Microsoft.NETCore.Targets/1.1.0": {},
      "Microsoft.Win32.Registry/4.6.0": {
        "dependencies": {
          "System.Security.AccessControl": "4.6.0",
          "System.Security.Principal.Windows": "4.6.0"
        }
      },
      "Newtonsoft.Json/13.0.1": {
        "runtime": {
          "lib/netstandard2.0/Newtonsoft.Json.dll": {
            "assemblyVersion": "13.0.0.0",
            "fileVersion": "13.0.1.25517"
          }
        }
      },
      "runtime.linux-arm.runtime.native.System.IO.Ports/4.6.0-rc2.19462.14": {
        "runtimeTargets": {
[... 286 lines cut ...]
      "hashPath": "system.security.accesscontrol.4.6.0.nupkg.sha512"
    },
    "System.Security.Principal.Windows/4.6.0": {
      "type": "package",
      "serviceable": true,
      "sha512": "sha512-Mdukseovp0YIGaz16FMH6nbfgZkrCFOJbtXQptv0aeBO9h775Ilb9+TDwLVTKikoW7y7CY7lpoXl9zmZ5G3ndA==",
      "path": "system.security.principal.windows/4.6.0",
      "hashPath": "system.security.principal.windows.4.6.0.nupkg.sha512"
    },
    "System.Text.Encoding/4.3.0": {
      "type": "package",
      "serviceable": true,
      "sha512": "sha512-BiIg+KWaSDOITze6jGQynxg64naAPtqGHBwDrLaCtixsa5bKiR8dpPOHA7ge3C0JJQizJE+sfkz1wV+BAKAYZw==",
      "path": "system.text.encoding/4.3.0",
      "hashPath": "system.text.encoding.4.3.0.nupkg.sha512"
    },
    "System.Threading.Tasks/4.3.0": {
      "type": "package",
      "serviceable": true,
      "sha512": "sha512-LbSxKEdOUhVe8BezB/9uOGGppt+nZf6e1VFyw6v3DN6lqitm0OSn2uXMOdtP0M3W4iMcqcivm2J6UgqiwwnXiA==",
      "path": "system.threading.tasks/4.3.0",
      "hashPath": "system.threading.tasks.4.3.0.nupkg.sha512"
    }
  }
}

ciramconsulting, 2022-02-11:
Fixed! You need to run dotnet ModbusApp.dll !!!! not ModbusApp.exe

closed 2022-02-11 by ciramconsulting

Apollo3zehn, 2022-02-11:
Good to know, thanks for your feedback!