Apollo3zehn/FluentModbus #66: Question : how should it be configured to act as Monitor/datalogger

valeriob, 2022-05-08:
Hi,
i would like to monitor the communication between a modbus master and it's slaves, is it possible to configure it to do so ?

Thanks

Apollo3zehn, 2022-05-09:
If you already have a Modbus master and slaves and this library is additional, I think you cannot use it for data logging because Modbus TCP uses 1 to 1 TCP connections. In case of Modbus RTU it would be easier to create a data logger but still problematic because this library has no "listen-only" mode. However, it would be easy to comment out the response logic and make this lib listen-only by yourself.

For a Modbus TCP setup, I would probably use Wireshark to monitor the communication.

Another solution could be to configure this library to act as slave AND master at the same time, i.e. by accepting incoming master requests and execute them on behalf of the master on the slaves and return the response back to the master. That would make logging easy but interfere with your current setup.

vborioni-onit, 2022-05-09:
Hi @Apollo3zehn thanks for you answer, you are right i forgot to specify that i'm referring to the serial RTU version :D 
Thank you for your suggestion, i'll look into the code ! when i will have some device to test i'll see if i can contribute back some config for readonly mode than !

valeriob, 2022-05-09:
Hi @Apollo3zehn thanks for you answer, you are right i forgot to specify that i'm referring to the serial RTU version :D 
Thank you for your suggestion, i'll look into the code ! when i will have some device to test i'll see if i can contribute back some config for readonly mode!

closed 2022-05-09 by valeriob