Apollo3zehn/FluentModbus #67: Implement Disposable

taai, 2022-05-15:
I suggest to implement [IDisposable](https://docs.microsoft.com/en-us/dotnet/api/system.idisposable). It is simpler to disconnect, if ```IDisposable``` is implemented.

When code gets complicated, you have two options:
* call ```Disconnect()``` in multiple places
* wrap all your code with try-finally and call ```Disconnect()``` only once in ```finally``` block

I offer implement this. Do you agree and should I make a PR? 😉 

Currently:
```csharp
var client = new ModbusTcpClient();
// code here ...
// disconnect
client.Disconnect();
```

Suggestion:
```csharp
using (var client = new ModbusTcpClient())
{
    // code here ...
    // will disconnect automatically
}
```

Apollo3zehn, 2022-05-15:
Good suggestion, please go ahead :-)

Apollo3zehn, 2022-09-08:
I has implemented it in v5, so I am closing this issue for now.

closed 2022-09-08 by Apollo3zehn