Apollo3zehn/FluentModbus #71: FluentModbus.ModbusException: 'The protocol identifier is invalid.'

kirillivan0ff, 2022-05-30:
Seems like a 
var unitIdentifier = 0xFF;
giving this error when I using your examples. 

The device manual saying:
"INPUT DATA AREA
The following table lists the registers of the input area (produced from the instrument and read by the
master), common to all PROFINET, ETHERCAT, ETHERNET/IP fieldbuses. The registers are 16 bit in size.
The input area is updated at a maximum frequency of 125 Hz (80 Hz in case of FIELBUS). The size
of the output area configured in the master fieldbus must match the size configured in the instrument"
Which seems to be OK. 16bit."

Any suggestions?

Apollo3zehn, 2022-05-30:
Is it a Modbus TCP or RTU problem? Can I reproduce it somehow?

From the error message it seems that the device is sending an incorrect response header but that is difficult to say.

kirillivan0ff, 2022-05-30:
> Is it a Modbus TCP or RTU problem? Can I reproduce it somehow?
> 
> From the error message it seems that the device is sending an incorrect response header but that is difficult to say.

It's TCP. Trying to speak with [this ](https://www.pavonesistemi.com/app_data/docs/products/manuals/mct-1302_technical_manual.pdf) guy and not really successful. Not sure if you can reproduce it but if I can check the header somehow I'll be glad to try it.

Apollo3zehn, 2022-05-30:
You could add a breakepoint to that line:

https://github.com/Apollo3zehn/FluentModbus/blob/ae88394c51cbe109c9ea5cae1c08bc57860fb6cc/src/FluentModbus/Client/ModbusTcpClient.cs#L251

And then see what data the routine received and compare it to the Modbus spec. Alternatively you could try Wireshark to capture the Modbus traffic.

kirillivan0ff, 2022-05-30:
It’s 8224 

Make any sense for you?

I can try to catch packets with Wireshark and this would probably be my next step… During next few days.

 

You could add a breakpoint to that line:

https://github.com/Apollo3zehn/FluentModbus/blob/ae88394c51cbe109c9ea5cae1c08bc57860fb6cc/src/FluentModbus/Client/ModbusTcpClient.cs#L251

Apollo3zehn, 2022-05-30:
8224 is not correct but I do not know why. I am sure Wireshark will help.

kirillivan0ff, 2022-05-31:
> 8224 is not correct but I do not know why. I am sure Wireshark will help.

OK, I think you can close this one for now. I will ping you up if I will find something interesting.

kirillivan0ff, 2022-06-01:
Hi, It’s me again.

The device reply with wrong Protocol ID. Is this number make some sense to you?

 

https://github.com/Apollo3zehn/FluentModbus/blob/ae88394c51cbe109c9ea5cae1c08bc57860fb6cc/src/FluentModbus/Client/ModbusTcpClient.cs#L251

Apollo3zehn, 2022-06-01:
You mean the value `8224`? No it does not really make sense to me. Do you have a Wireshark screenshot or log of the full response of your device?

karateboy, 2024-02-08:
I ran into the same situation. 
I also take the aysnc/await usage. 
I use it for device calibration. In this case, dely during between reading/writing is very common practice.

karateboy, 2024-02-08:
@Apollo3zehn Please mark this issue as open. I may take sometime study how to fix it.

karateboy, 2024-02-08:
It is common to use a modbus gateway to map Modbus RTU devices into Modbus TCP registers.
Since they are in fact different devices, my application are using different tasks with individual modbus clients to read/write those Modbus registers/coils. 

I found that without locking the read/write may fail randomly. (after enforcing SemaphoreSlim locking, this kind of error disappear) I don't think it a bug. However, if the library can maintain a semaphore for the modbus device (ID is IP + slave ID), it can save a lot of time for people like me to figure out what went wrong.

As for the 'The protocol identifier is invalid.' error, I suspect it may result from my "configureAwit(false)". I just chain all my assync modbus actions into "configureAwit(true)" to see if the error disappear. It may take some time to test.

Apollo3zehn, 2024-02-21:
It would be great if you create a pull request or show me some code to illustrate the required changes :-) Thank you

karateboy, 2024-02-25:
I wish I can provide you a pull request but I don't have time to prepare it and fully tested. 
So instead, I provide you my workaround for youre reference.

I hava a wrapper class called **ModbusClient** which encapsulate your **ModbusTcpClient**.
**ModbusClient** has a static ConcurrentDictionary storing pairs of semaphore and IP/Port and a internal semaphore reference.
public class ModbusClient : IDisposable
{
    private record ModbusTarget(IPAddress Address, int Port);
    private static readonly ConcurrentDictionary<ModbusTarget, SemaphoreSlim> TargetLockMap = new();
    private readonly SemaphoreSlim _targetLock;
..
}

In the constructor, It try to get/add the semaphore from/to the concurrent map. Thus, for modbus clients of the same IP, they will share the same semaphore.
public ModbusClient(Device device...)
 {
...
        var modbusTarget = GetModbusTarget(device); // Generate record according to device IP and port
        TargetLockMap.TryGetValue(modbusTarget, out var targetLock);
        _targetLock = targetLock ?? TargetLockMap.GetOrAdd(modbusTarget, new SemaphoreSlim(1, 1));
 }

For each modbus read/write operation, e.g. read coil. **ModbusClient**  has to wait for the semaphore before the operation.
public async Task<List<(DeviceSignalIo.IDeviceSignal, bool)>> ReadSignalAsync(CancellationToken cancellationToken)
    {
        try
        {
            await _targetLock.WaitAsync(cancellationToken).ConfigureAwait(false);
            ...
            cancellationToken.ThrowIfCancellationRequested();

             var coilMemory =
                        await client.ReadCoilsAsync((byte)device.SlaveId, (ushort)signal.Address + signal.Offset, 1,
                            cancellationToken);
        }
        catch (Exception ex)
        {

        }
        finally
        {
            _targetLock.Release();
        }

        return recordList;
    }

For async task, it is possible different tasks executing on the same thread may read/write modbus master at the same time. Without a semaphore, multiple read/write may try to access the underlying network stream at the same time. A Semaphore(1, 1) ensure only one operaton can be performed until the previous result come back. 

In contrast, for disk or other high throughput devices, they support outstanding I/O, which means the master can issue multiple requests before the previous result return. However, for modbus devices, this kind of operation is not supported.

karateboy, 2024-02-25:
Here is my scenario, I hava multiple timely tasks to reading modbus devies data at the same time. Also I have a task to perform the calibration of the modbus devices. Each task has its onw ModbusClient and may operate at the same time.
By the way, after I add those logics, the protocol identifier invalid disappear.