Apollo3zehn/FluentModbus #76: [Question] Is the frame delay accounted for in this library

horeich, 2022-08-02:
Hello,

I'm currently using this library to read data from a client. Works well so far. Thank you!

Just one thing: Is the implementation according to the specification with a 3.5 char delay or/and a minimum interframe time? I'm asking because I require the highest possible performance and a fixed interframe time would be too much of a delay in my use case.

Thank you,
Andy

Apollo3zehn, 2022-08-02:
Actually there is not minimum delay between frames enforced by FluentModbus since I was not aware of such a requirement. I will add it according to the spec. So thanks for mentioning it!

I guess no problems occurred so far because FluentModbus (which uses the 
NET SerialPort) is slow enough to always fulfill the requirement of 3.5 character times between frames).

Do I understand you right that you do not want any fixed delay (as it is the case right now)? So when I implement it, you need a way to switch it off?

horeich, 2022-08-03:
Thanks for your answer.

Right, for my application I require the maximum throughput which, according to the specification, would not be possible. So yes it should be able to be switched off.

The interframe delay is not very efficient as it will decrease up to to a certain baud rate (obviously 3.5 chars between each frame  have different length for different baud rates) only to be a fixed value above a certain baud rate (115200?). See p. 14 [Modbus Spec](https://modbus.org/docs/Modbus_over_serial_line_V1_02.pdf)

Apollo3zehn, 2022-08-03:
Okay, then I will implement it and make it possible to switch it off. I am on holidays right now so I`ll do it when I am back.

horeich, 2022-08-03:
Thank you @Apollo3zehn, enjoy your vacations!

Apollo3zehn, 2022-08-16:
I will implement it later, no time right now and I am not sure how to properly implement it. The following link is only for myself for later reference: https://stackoverflow.com/questions/20740012/calculating-modbus-rtu-3-5-character-time#21223624