Apollo3zehn/FluentModbus #82: NetworkStream disposed

manuelelucchi, 2022-10-17:
Hi, we are using FluentModbus to connect 24/24 7/7 on a machine. The problem is, after some days of works the library fails with the NetworkStream disposed. Do you have any idea on when this could happen? I have no further informations as for now since it's difficult to reproduce, we'll try with continuous connections and disconnections to avoid the object being disposed

Apollo3zehn, 2022-10-18:
Hi, maybe a stack trace could help finding the cause of the issue. Disposing should only happen if the connection times out, I do not think that there is an `OutOfMemoryException` or a stack overflow that could also maybe lead to disposed network stream.

Which version of FluentModbus do you use and how are you working with the client? Are you disposing the client maybe accidentially?

manuelelucchi, 2022-10-20:
I'm trying to get the stacktrace. Apart from this, the usage is something like this

```csharp
while (!token.IsCancellationRequested)
{
	while (!client.IsConnected)
	{
		try
		{
			client.Connect();
		}
		catch (Exception e)
		{
			// log
		}

		await Task.Delay(Configuration.Polling, token);
	}

	try
	{
		// do stuff
	}
	catch (Exception e)
	{
		// log
	}

	await Task.Delay(Configuration.Polling, token);

	resetCounter++;
	if (resetCounter== resetMax)
	{
		resetCounter= 0;
		client.Disconnect();
	}			
}

if (client.IsConnected)
{
	client.Disconnect();
}

client.Dispose();
```

And, since the error is repeated continuously, there should not be accidental disposing I guess

manuelelucchi, 2022-11-21:
Hi @Apollo3zehn, we managed to corner the issue a bit more. After some tweaking, it's now a NullReferenceException inside the ModbusTcpClient while awaiting ReadHoldingRegistersAsync<short>(unit, address, count, token)
```
in FluentModbus.ModbusTcpClient.<TransceiveFrameAsync>d__40.MoveNext()
   in FluentModbus.ModbusClient.<ReadHoldingRegistersAsync>d__30.MoveNext()
   in FluentModbus.ModbusClient.<ReadHoldingRegistersAsync>d__29`1.MoveNext()
   in MyNamespace.MyClass.MyMethod.<ReadStandardMatrixAsync>d__7.MoveNext() in
```

where the parameters are:
- unit: 1
- startingAddress: 1000
- count: 100

and the session properties are:
- IsConnected: true
- ConnectTimeout: 1000
- ReadTimeout: -1
- WriteTimeout: -1

This happens really randomly so it's really difficult to reproduce

sharpbetter, 2023-03-08:
hi, when I quickly switch between server-side connection states, this problem may occur
i think, `IsConnected`  check  `TcpClient`   connected
when `TcpClient` connection is ready , but `_networkStream`not ready , 

[image]