Apollo3zehn/FluentModbus #83: DetectResponseFrame failed

zhangchaoza, 2022-11-15:
ModbusClient call WriteSingleCoilAsync or WriteSingleRegisterAsync timeout when registerAddress over 1023.  Test code as as follows：

```csharp
var client = new ModbusRtuClient()
{
    BaudRate = 115200,
    Parity = System.IO.Ports.Parity.None,
    StopBits = System.IO.Ports.StopBits.One,
    ReadTimeout = 3000,
};
client.Connect("COM7", ModbusEndianness.BigEndian);
await client.WriteSingleCoilAsync(1, 0x03FF, true); // write Coil address of 1023
await client.WriteSingleCoilAsync(1, 0x0400, true); // write Coil address of 1024 timeout
await client.WriteSingleRegisterAsync(1, 0x03FF, 1);// write reg address of 1023
await client.WriteSingleRegisterAsync(1, 0x0400, 1);// write reg address of 1024 timeout
```

output:

```log
Unhandled exception. System.TimeoutException: The asynchronous read operation timed out.
   at FluentModbus.ModbusRtuSerialPort.ReadAsync(Byte[] buffer, Int32 offset, Int32 count, CancellationToken token) in /home/runner/work/FluentModbus/FluentModbus/src/FluentModbus/ModbusRtuSerialPort.cs:line 107
   at FluentModbus.ModbusRtuClient.TransceiveFrameAsync(Byte unitIdentifier, ModbusFunctionCode functionCode, Action`1 extendFrame, CancellationToken cancellationToken) in /home/runner/work/FluentModbus/FluentModbus/src/FluentModbus/Client/ModbusRtuClientAsync.cs:line 58
   at FluentModbus.ModbusClient.WriteSingleCoilAsync(Int32 unitIdentifier, Int32 registerAddress, Boolean value, CancellationToken cancellationToken) in /home/runner/work/FluentModbus/FluentModbus/src/FluentModbus/Client/ModbusClientAsync.cs:line 267
```

And the Modbus Slave software logs as follows：

```
Rx:000145-01 05 03 FF FF 00 BC 4E 
Tx:000146-01 05 03 FF FF 00 BC 4E 
Rx:000147-01 05 04 00 FF 00 8D 0A 
Tx:000148-01 05 04 00 FF 00 8D 0A 
Rx:000157-01 06 03 FF 00 01 78 7E 
Tx:000158-01 06 03 FF 00 01 78 7E 
Rx:000159-01 06 04 00 00 01 49 3A 
Tx:000160-01 06 04 00 00 01 49 3A 
```

I found that  DetectResponseFrame method will always return false when detect `01 05 04 00 FF 00 8D 0A`.

Apollo3zehn, 2022-11-17:
Thank you - the DetectResponse frame implementation seems to be incorrect for write methods. I´ll publish an update now.

Apollo3zehn, 2022-11-17:
Should be solved now with v5.0.2. Unfortunately I still have not good Modbus RTU test setup so I would be glad if you can confirm that the bug does not appear any longer.

closed 2022-11-17 by Apollo3zehn

zhangchaoza, 2022-11-18:
Yes it works fine and thank you

JohnMasen, 2022-11-20:
just about to raise this issue.  
After upgrade to 5.0.2, I ran my tests and  it is confirmed that the issue is fixed. 
Thank you guys, very appeaciated for your work.

commit 24317f58 (2022-11-17): Fix #83.