Apollo3zehn/FluentModbus #87: [Question] - OpenPLC integration with esp32

ghostidentity, 2023-02-12:
Hello Guys,

I noticed that OpenPLC can take advantage of esp32 to establish connection to modbus server. 
I wanted to have a scenario wherein:

1. The user press the push button (esp32 device) 
2.  The device then send request to c# program (server) that integrates with fluentmodbus e.g. pin 2 (high)
3. C# program acknowledge the request then modify the pin value to esp32device

The problem is that, openPLC does not appear to connect to c# program (server mode), so I'd like to ask if it's possible ?

[image]

Thanks,
Mark

Apollo3zehn, 2023-02-12:
Hi, I have a few questions:

- Why do you use 127.0.0.1 as IP address? Is your OpenPLC running on the same computer as the Modbus server?

- What does "... that integrates with fluentmodbus e.g. pin 2 (high)" mean? I don't understand how a microcontroller pin is related to Modbus TCP.

- How should the Modbus TCP server react to the request? You write it should modify the pin value of the ESP device. But I do not understand how that would be possible if the server runs on a different device.

- How would a simple TCP Connect help to communicate via Modbus TCP? Where is the rest of the protocol?

I think my main problem is that I do not understand your architecture. Could you please create a diagram/text which explains your setup in more detail?

If you have network problems, I always recommend Wireshark. It would show you if the TCP Connect was successful. It also recognizes Modbus communication.

ghostidentity, 2023-02-13:
Here's the general overview. 
[image]

1. Yes, OpenPLC can be simulated locally, since I executed openplc and modbus server on same computer the address  127.0.0.1 can be used but it code is deployed on esp32 device, the address will be changed.
2. On the diagram, when the person click the push button it will send a data to modbus server like: device_1: true
3.  Once the device successfully connected to the modbus server, technically the server can write coil values or the server can connect to specific device to modify the coil position [2] value to 1.  Based on openPLC mapping,  position [2] = pin #2, setting it to 1 (high) will turn on the led light.
4.  How would a simple TCP Connect help to communicate via Modbus TCP? Im not sure about this one, need to understand how it connect, as its OpenPLC related. But I tried the sample code provided on fluentmodbus documentation, i was able to read the coil value using Modbus TCP client. So my assumption is that, I can use the ModBusTCPServer so the device can connect to it.

Other info:

OpenPLC, can configure the device to use ModBus TCP for communication.

[image]

I will try wireshare to troubleshoot.

Apollo3zehn, 2023-02-13:
Thanks, so my understanding is:

- button is connected to esp32
- esp32 detects if button is pushed
- then it sends some data ("device_1: true") to the Modbus TCP server via WiFi (note: sending string data via Modbus is maybe not the best idea. Or is it just a sample and the esp32 just sends anything to somehow notify its button has been pushed? You can easily send numeric data like int32 or single bits (coils).)

- What happens then? Your write the server can write coil values ... yes, but only to its internal storage, not on a different device. The Modbus TCP server is passive. Only the Modbus client can actively read or write data from/to a server. 

For a true bidirectional communication, your would need a client and server on both sides, esp32 and on the server device.

Another solution would be to let the esp32 poll the server for changes. The server cannot push data.

Do I get it right so far? Or does "the server can connect to a specific device" mean using a different protocol other than Modbus?