Apollo3zehn/FluentModbus #88: ModbusUtils as public class

lerppana, 2023-02-24:
Hi! I'm making something with your library. I have a case in which I would like to call `ModbusUtils.SwitchEndianness(Span<T>)` from my client code, however class seems to be internal. Is there a case for not making it public? Would you mind making it public and making new (pre?)release?

Apollo3zehn, 2023-03-01:
Hi, could you please shortly describe your use case? Why do you need access to the internal class? I would just like to understand if the API of the library is insufficient or if you are maybe using it wrong (caused by bad documentation :-)). Thank you!

lerppana, 2023-03-08:
Hi! I'm perhaps missusing your library a bit 😅 I have couple of datatypes that I need to read as plain bytes and do some extra stuff with them. For debugging purposes I was reimplementing the `ReadHoldingRegisters<T>` so that I 
- ReadHoldingRegisters (plain bytes) 
- debug print (not really necessary but helpful when I started)
1. If datatype is normal then just MemoryMarshal and SwitchEndianness if required
2. For special types do their own logic
- Return from same function

What I was thinking was having these things in same code path. What I have now is two paths for normal types where I call `ReadHoldingRegisters<T>` and other for special byte tinkering after plain `ReadHoldingRegisters`. This is perfectly fine but for clean code reasons  I opted into one code path and copy pasted the SwitchEndianness function from your codebase into mine. This is not really optimal way to do it, I know 😅