Apollo3zehn/FluentModbus #94: 单个读会报空引用错误

PengYouLin, 2023-07-12:
我用docker部署了一个client端，modbus的IP地址给的是127.0.0.1.容器默认的IP地址。
但是这个_networkStream变量为null。后续在使用的时候，没有判断该变量是否为空，返回报错：Object reference not set to an instance of an object.
[image]

Apollo3zehn, 2023-07-17:
How do you initialize and connect your Modbus TCP client? Tha would help me to understand how the stream can be null.

closed 2023-08-03 by Apollo3zehn

PengYouLin, 2023-08-04:
[image]
我的网络环境不是很好，会存在时断时续的情况，
当我根据isconncted为false的时候，会重新new 一个client对象，
这个时候会重新传进去一个new tcpclient.
看你的源码，是在new对象的时候，就给networkstram赋值了，后续读的时候，都是使用这个networkstram。如果这个流对象为null，则write和read时都会报空引用。