Apollo3zehn/FluentModbus #99: RTU over TCP

puschie286, 2023-09-29:
Hi, do you plan to add RTU over TCP support ? its the last thing we need to replace other packages with fluentmodbus
( [some informations about it](https://github.com/NModbus/NModbus/issues/5) )

Apollo3zehn, 2024-02-21:
I do not have much spare time for this project so a pull request would help to speed up bringing this features :-)

Apollo3zehn, 2024-08-08:
I have never used RTU over TCP so I cannot give you any input about this. But it would be great if you work on a PR to add this feature!

schotime, 2024-08-22:
@Apollo3zehn just in case you didn't see, but I submitted the PR, and it's been running for a couple weeks now from a local build. Let me know if there is anything else you need to me to change

Apollo3zehn, 2024-08-23:
Thank you, I`ll check it today

Apollo3zehn, 2024-09-19:
Closed by #126

closed 2024-09-19 by Apollo3zehn