Apollo3zehn/FluentModbus #102: SampleServerClientTCP broken?

raymondbh, 2023-12-01:
Just testing your modbus library (that looks very good at first glance), but I get just get "TCP connection closed unexpectedly" what ever I do. Tried two different computers and added firewall rules and everything I can think of...

[image]

SampleServerClientRTU seems to work just fine :)

raymondbh, 2023-12-01:
Ok, I figured it out, seems the client unitiIdentifier set to 0xFF in the sample does not work. I set it to 0 and it was fine :) 

//var unitIdentifier = 0xFF;
var unitIdentifier = 0x00;

[image]

Apollo3zehn, 2024-02-21:
Thank you for pointing that out, I just fixed it

commit b7bab82e (2024-02-21): Prepare release and fix #102
closed 2024-02-21 by Apollo3zehn