Apollo3zehn/FluentModbus #106: ModbusRTUclient ReadHoldingRegister error System.TimeoutException: 'The operation has timed out.'

bobfri, 2024-01-19:
if i only read a register once i don't get any issue but when i try to read for a second time i get the error.
this is my current test code:

  static void Main(string[] args){
          Console.WriteLine("Hello, World!");
          ModbusRtuClient m = new ModbusRtuClient()
          {
              BaudRate = 115200,
              Parity = System.IO.Ports.Parity.None,
          };
          m.Connect("COM7");
          var t = m.ReadHoldingRegisters(1, 0x0070, 2);
          var tt = m.ReadHoldingRegisters(1, 0x0070, 2);
   }
}

Apollo3zehn, 2024-01-20:
I am sorry, I have no RS232 hardware available and Modbus RTU implementation was tested once via a virtual COM bridge. Is it working with other Modbus RTU tools? I suspect the problem being in the device but I cannot be sure.

bobfri, 2024-01-22:
My setup runs RS485 and i have managed to get it working with sevral other Modbus RTU tools so i dont think its a device issue.