Apollo3zehn/FluentModbus #108: client.WriteMultipleRegisters. Write error

Seba131, 2024-02-04:
Hi.
I have a problem with:
`client.WriteMultipleRegisters(unitIdentifier, startingAddress, new byte[] { 10, 00, 20, 00, 30, 00, 255, 00, 255, 01 });` 
shows me errors:
"Specified argument was out of the range of valid values"  or "Object reference not set to an instance of an object"

ReadHoldingRegisters it works without any problems, so all settings are ok
`data = client.ReadHoldingRegisters<byte>(unitIdentifier, startingAddress, 10);`

Apollo3zehn, 2024-02-05:
Could you please also post the stack trace? Thanks :-)

Seba131, 2024-02-05:
> Czy mógłbyś również opublikować ślad stosu? Dzięki :-)

Ok, my error in the code works. I implemented BaudRate and Parity incorrectly. ModBus transmission is well written, I RECOMMEND.
Thank you for your interest in my problem.

closed 2024-02-05 by Seba131