Apollo3zehn/FluentModbus #114: Signals, RTS, DTR, CTS ???

Seba131, 2024-03-15:
Good morning, is there a chance to add support for RTS, DTR, CTS signals? They would be very helpful to me.

Apollo3zehn, 2024-03-15:
Hi, you can use your self-managed SerialPort by doing the following:

```cs
var serialPort = new SerialPort("COM1")
{
    DtrEnable = true,
    RtsEnable = true
};

serialPort.Open();

client.Initialize(
    new ModbusRtuSerialPort(serialPort), 
    ModbusEndianness.LittleEndian);
```

Seba131, 2024-03-16:
```
public ModbusRtuClient modbusRtu = new ModbusRtuClient();

modbusRtu.BaudRate = 9600;
itp...
```
No option to select DTR, CTS....

[image]

Apollo3zehn, 2024-03-17:
I am sorry, my example above missed the creation of the `ModbusRtuClient`:

Here is a better sample now:

```cs
var serialPort = new SerialPort("COM1");
var client = new ModbusRtuClient();

client.Initialize(
    new ModbusRtuSerialPort(serialPort), 
    ModbusEndianness.LittleEndian);

serialPort.Open();

serialPort.DtrEnable = true;
serialPort.RtsEnable = true;
```

As long as you have control over the instantiation of `ModbusRtuClient` the code sample should work. Or do you encounter other problems I did no cover in my answer (I did not fully understand your last post so I am not sure)?

Seba131, 2024-03-20:
ok. It works.

closed 2024-05-31 by Apollo3zehn