Apollo3zehn/FluentModbus #116: ModbusTCPServer view connected clients

GREEFA-gvanos, 2024-05-03:
Hello,

First of all i'm using the library and think it is a great implementation.
However i would like to have the following posibilities:

- View client IPAdresses
- View client connection status

Currently it is possible to view the amount of connected clients via the `ConnectionCount` Property it would be nice to also see for each connected client what the IP address of that client is.

Maybe an event on the client connection is also good enought. for examlpe:
- Client connects
- Event is raised with paramter that contains IPAddress.

Please let me know if something like this is possible or planned for the future.

Greetings
Gerrit van Os

GREEFA-gvanos, 2024-05-17:
After looking into the ITCPClientProvider interface i could build my own implementation for this and add the connected client features i was asking for so this issue is not relevant anymore

closed 2024-05-17 by GREEFA-gvanos

Apollo3zehn, 2024-05-17:
Sorry for not answering yet. Glad you found a solution.