Apollo3zehn/FluentModbus #119: Stackoverflow Exception when I try to reconnect to a modbus TCP server

epasinetti, 2024-05-29:
After many times a modbus server not respond the ModbusTcpClient crash and give me StackOverflow Exception.
Any solutions for this issue ? 

Thank you

epasinetti, 2024-05-30:
Can you help me? ... I don't know how to get "stack trace"

Apollo3zehn, 2024-05-31:
The full exception message includes the stack trace which tells us in which line of code the original exception occurred.

epasinetti, 2024-05-31:
[image]

Apollo3zehn, 2024-05-31:
Thank you! It looks like the calling function `twModbus.Init()` or `twModbus.Connect()` is recursively calling itself, problaby once per failed connection attempt and after a while the stack overflows. Can you confirm the calling function recursively calls itself (or one of its parents)? I don't think it is a problem of FluentModbus actually.

epasinetti, 2024-06-01:
Thank you ! the problem was in my code !

closed 2024-06-01 by Apollo3zehn