Apollo3zehn/FluentModbus #120: How to reverse the order of each Input registers

MonoLogueChi, 2024-06-12:
Sorry to bother you, I have encountered doubts when using modbus. I don't know how to get the correct value.

I need to read a float value, I use the following code

```cs
var axis0 = new ModbusTcpClient();
axis0.Connect(new IPEndPoint(IPAddress.Parse("192.168.0.233"), 502), ModbusEndianness.LittleEndian);
var a = await axis0.ReadInputRegistersAsync(1, 0, 2);
```

And I got the data

```
0x09 0x37 0x42 0x48
```

But I need the correct data to be

```
0x37, 0x09, 0x48, 0x42
```

I can only do a manual conversion now. Does FluentModbus have an API that can do automatic conversion?

Apollo3zehn, 2024-06-13:
Hi, you could use `ModbusEndianess.BigEndian` and read `ushort` data:

```
axis0.Connect(new IPEndPoint(IPAddress.Parse("192.168.0.233"), 502), ModbusEndianness.BigEndian);
var a = await axis0.ReadInputRegistersAsync<ushort>(1, 0, 2);
```

In that case you should get two `ushort` values with the correct byte order. Here is a bit more information:
https://github.com/Apollo3zehn/FluentModbus/issues/86#issuecomment-2111742545

MonoLogueChi, 2024-06-13:
> Hi, you could use `ModbusEndianess.BigEndian` and read `ushort` data:
> 
> ```
> axis0.Connect(new IPEndPoint(IPAddress.Parse("192.168.0.233"), 502), ModbusEndianness.BigEndian);
> var a = await axis0.ReadInputRegistersAsync<ushort>(1, 0, 2);
> ```
> 
> In that case you should get two `ushort` values with the correct byte order. Here is a bit more information: [#86 (comment)](https://github.com/Apollo3zehn/FluentModbus/issues/86#issuecomment-2111742545)

Thank you very much

closed 2024-06-13 by MonoLogueChi