Apollo3zehn/FluentModbus #121: Errors with Different Clients on FluentModbus Server

rligocki, 2024-06-30:
Description

I created a Modbus server using the FluentModbus library for .NET. Initially, I used the FluentModbus client for reading and writing to the server without any issues. However, upon switching to different types of Modbus clients, I encountered several errors during read and write operations.

Steps to Reproduce

Set up a Modbus server using the FluentModbus library for .NET.
Use the FluentModbus client to read and write to the server - no issues observed.
Switch to other Modbus clients:
ModbusTcpClient (macOS, available on Apple App Store):
Error: Network Connection Failed POSIXErrorCode(rawValue: 54): Connection reset by peer
Golang Modbus library by simonvetter ([GitHub link](https://github.com/simonvetter/modbus)):
Errors:

```
WARN[2024-06-30T20:26:30.739425+02:00] Failed to write to Modbus: EOF
WARN[2024-06-30T20:26:31.739785+02:00] Failed to close connection to Modbus: %!s(<nil>)
```

Expected Behavior

The server should handle read and write requests from different Modbus clients without errors.

Actual Behavior

Errors are encountered when using ModbusTcpClient and the Golang Modbus library, whereas the FluentModbus client works without issues.

Additional Information

FluentModbus version: 5.2.0
.NET version: 8.0
Operating System: macOS Sonoma 14.4.1

Questions

Could this be an issue with the FluentModbus server implementation?
Is there a compatibility issue with non-FluentModbus clients?
Any insights or suggestions to resolve these errors would be greatly appreciated.

Apollo3zehn, 2024-07-01:
`Connection reset by peer` seems to be a configuration error, either of your firewall or of the Modbus TCP server. How are you starting the server. Are you giving it any IP address to listen on, or are you using the defaults?

rligocki, 2024-07-01:
Connection was created on localhost with firewall disabled. For now more important error for me is EOF. 

Start of unit modbus server:
`newUnit.Server.Start(new IPEndPoint(IPAddress.Parse("0.0.0.0"), port: _unitStartPort+i));`

Apollo3zehn, 2024-07-01:
The reason for "EOF" is that the go client by default requests data from a server with unit identifier = 1. But FluentModbus by default accepts messages with unit identifier = 0 which is in compliance with the spec:

[image]

In contrast, go client seems to have `1` as its default: 
[image]

With the following CLI arguments I was able to change the unit identifier to `0` and then read the data:

```json
"args": [
    "--target",
    "tcp://localhost:1999",
    "--timeout=3s",
    "sid:0",
    "rh:uint16:0x000+5"
]
```

During testing I got the log message in FluentModbus that the client connection timed out which was not true. Actually the connection was closed due to an exception. The log output will be improved in future so that it looks more like this:

[image]

Apollo3zehn, 2024-07-01:
Alternatively you can add the unit `1` to the `FluentModbus` server via `server.AddUnit(1);`

rligocki, 2024-07-02:
Thank you. Will try that fix.

closed 2024-07-22 by Apollo3zehn