Apollo3zehn/FluentModbus #125: Should swap bytes be an extension Method to span? One method per swap type (little, big, vax endian)  ... How to handle type size?

Apollo3zehn, 2024-08-07:
https://github.com/search?q=repo%3AApollo3zehn%2FFluentModbus%20swapbytes&type=code

iamalexcater, 2024-08-07:
My gut feeling is no; would it not be better to follow the design of the BinaryPrimitives class? (or use it, ultimately)

Apollo3zehn, 2024-08-08:
I will work out some use cases and then try to make all of them usable in a clear way. But thanks for your feedback!

I have often tried to use `BinaryPrimitives` but for high performance I want to use a vectorized algorithm (which already exists in a [different project](https://github.com/Apollo3zehn/PureHDF/blob/dev/src/PureHDF/Filters/EndiannessConverterAvx2.cs)). I just need to copy it over.