Apollo3zehn/FluentModbus #133: Pass new value register value on write function codes to request validator

AthenaAzuraeaX, 2024-10-02:
The request validation function today takes Unit identifier, Function code, Address and Quantity of registers.

Do you think it would be possible to also pass the new register value from a write function code? That would allow us to validate the incoming data and possibly return Illegal Data Value exception code if it fails.

Apollo3zehn, 2024-10-10:
If that is doable in a performant way, I have nothing against it. Maybe it should be a flag (default off) so it does not affect performance if not needed. It would be a bad idea to create a new array, i.e. a new object the garbage collector has to track, for every single request that goes into the server. But I see that there might be the need for it. Could you draft a PR? It can be a breaking change because I am preparing v6 right now.

Apollo3zehn, 2024-10-10:
#134 describes a similar issue and the solution proposed there (a new type for the request validator args) is something I like. So if you consider creating a PR please base the implementation on that branch. Thanks :-)

AthenaAzuraeaX, 2024-10-14:
I will see if I can find the time to submit a PR for it. I believe it should be doable in a performant way using spans.