Apollo3zehn/FluentModbus #136: [Question] Whether the server can connect to multiple clients and separate control client

DYZ-H, 2024-11-21:
[image]
In sample code ,I can not find a way to change data with an unitIdentifier,but client will with an unitIdentifier when change data.
I wonder how server  to connect multiple clients and control client separately

Apollo3zehn, 2024-11-22:
Instead of `var registers = server.GetHoldingRegisters();` you can write `var registers = server.GetHoldingRegisters(unitIdentifier: xxx);`

And on the server side you can create more units and control them via:

[image]

DYZ-H, 2024-11-25:
Thank you very much for your reply,I have tested it ss you said，works fine。

closed 2024-11-25 by DYZ-H