Apollo3zehn/FluentModbus #137: I need to keep a log of the original communication, can I do it?

watking, 2024-11-22:
This project is great, I want to integrate it into my project, but I need to keep all the incoming and feedback data streams, can I do that? Achieve the same effect as the image
[image]

Apollo3zehn, 2024-11-22:
This is slightly related to #133.

I see the need for this but from a performance perspective (and this library is for high performance), it is difficult to implement. 

Maybe there should be a global switch to enable this feature and if it is enabled, an event is raised with the frame buffer as parameter. Something like this but for the client side:

[image]

For received data, this event would need to be raised right before this line:

https://github.com/Apollo3zehn/FluentModbus/blob/f96247576725ec9dcd764d5bd4f8c5402f2b56a1/src/FluentModbus/Client/ModbusTcpClient.cs#L311

And it would look like this:

```cs
if (EnableRaisingEvents)
    FrameReceived?.Invoke(this, new FrameReceivedEventArgs(frameBuffer.Buffer));

// ...
```

And then something similar for the frame data to be sent which would be right before this line:
https://github.com/Apollo3zehn/FluentModbus/blob/f96247576725ec9dcd764d5bd4f8c5402f2b56a1/src/FluentModbus/Client/ModbusTcpClient.cs#L239

So if you have time I would happily accept a merge request. Unfortunately I am busy with other projects right now so I cannot do it myself :-(

watking, 2024-11-23:
Thank you very much for your reply, I'm willing to try it, this feature can be used for data review