Apollo3zehn/FluentModbus #145: Serial Port not found

MassimilianoScarpetta, 2025-03-28:
Hi @Apollo3zehn!

First of all thank you for your awesome work, this library is very performant and easy to use.
Anyway, this time, I got an issue while trying to connect to a serial port.

I'm using a nport 5110 device that converts ethernet-RS485 signals. It also maps and extends serial ports.
I'm using your library to try to connect to the serial port that the nport device created.
Unfortunately the library don't find the serial port. I've tried to use ModbusPoll to connect to the port and it worked.

Could you please tell me if I misunderstood something or if there can be a workaround to solve this?

Apollo3zehn, 2025-03-28:
`Fluentmodbus` uses the .NET package `System.IO.Ports` to interact with COM ports. And I think the quickest solution for you would be to try to connect to your Serial port directly via this library. Maybe the COM port has a different name? You can list all names via `SerialPort.GetPortNames`: https://learn.microsoft.com/de-de/dotnet/api/system.io.ports.serialport.getportnames?view=net-9.0-pp&viewFallbackFrom=dotnet-plat-ext-8.0

```cs
using System;
using System.IO.Ports;

namespace SerialPortExample
{
    class SerialPortExample
    {
        public static void Main()
        {
            // Get a list of serial port names.
            string[] ports = SerialPort.GetPortNames();

            Console.WriteLine("The following serial ports were found:");

            // Display each port name to the console.
            foreach(string port in ports)
            {
                Console.WriteLine(port);
            }

            Console.ReadLine();
        }
    }
}
```

I hope this helps!

MassimilianoScarpetta, 2025-03-28:
Thank you for the reply!
Unfortunately using GetPortNames() method does not work because it seems that returns only "Real" ports and not virtual ports.
The MOXA device creates virtual ports that I'm trying to use.
Does your library works also on virtual ports? Maybe the issue is the number of the COM Port (in my case or 121 or 371)?

Apollo3zehn, 2025-03-28:
The library works with whatever `System.IO.Port` offers. But you can pass your own serial port by implementing `IModbusRtuSerialPort`. So maybe you are able to create your own serial port via P/Invoke and the Windows API.

You can then use it as shown here: 

https://github.com/Apollo3zehn/FluentModbus/blob/bbd7dcd1d628811a73328233700ca352b2d3d71c/src/FluentModbus/Client/ModbusRtuClient.cs#L101-L109